oxidecomputer/humility #57: Removal of submodules broke cargo install

iangrunert, 2021-12-09:
On latest master, I get build failures when running `cargo install --path .` (and hence on the recommended install method of `cargo install --git ssh://git@github.com/oxidecomputer/humility.git`).

```
error[E0308]: mismatched types
   --> cmd\spd\src\lib.rs:140:42
    |
140 |     let dev = spd::Function::PageAddress(page).to_code().unwrap();
    |                                          ^^^^ expected struct `Page`, found `u8`

error[E0599]: no method named `to_code` found for enum `Function` in the current scope
   --> cmd\spd\src\lib.rs:140:48
    |
140 |     let dev = spd::Function::PageAddress(page).to_code().unwrap();
    |                                                ^^^^^^^ method not found in `Function`

error[E0599]: no method named `to_code` found for enum `Function` in the current scope
   --> cmd\spd\src\lib.rs:258:55
    |
258 |         ops.push(Op::Push(spd::Function::Memory(addr).to_code().unwrap()));
    |                                                       ^^^^^^^ method not found in `Function`

error[E0599]: no method named `to_code` found for enum `Function` in the current scope
   --> cmd\spd\src\lib.rs:281:51
    |
281 |             let dev = spd::Function::Memory(addr).to_code().unwrap();
    |                                                   ^^^^^^^ method not found in `Function`

Some errors have detailed explanations: E0308, E0599.
For more information about an error, try `rustc --explain E0308`.
```

This appears to compile under 68df3a17b6f4f52e211f40287d175239fb1a9f53 which is strange, not sure how the submodule removal in #56 could have broken this?

steveklabnik, 2021-12-10:
I believe this is a consequence of https://github.com/rust-lang/cargo/issues/7169; that is, changes in the spd crate are not reflected here, because we are https://github.com/oxidecomputer/humility/blob/master/Cargo.lock#L1594 at `f56bca9fbc380355cfe904912bf73789abd06e26` but `HEAD` of spd is `e37e79f6d7d4805b8a6a8c4d37699c4bd60222ea`.

I believe if you pass `--locked` to `cargo install`, it will work. Is that the case?

tarka, 2021-12-11:
`--locked` worked.

iangrunert, 2021-12-11:
Yep, `--locked` worked, thanks. I'll make a PR to hubris to update the instructions for installing humility in the README there.

closed 2021-12-11 by iangrunert