oxidecomputer/humility #181: USB linking still broken

rmustacc, 2022-07-05:
I recently tried to update humility and as trying to build it. It failed linking the final binary due to undefined symbols. This was recently cited as being due to having older versions of libusb-0.1-4 and ensured that a libusb-1.0-0 variant was installed (libusb-1.0-0/focal,now 2:1.0.23-2build1). It's not clear to me what the next bit is there. Environment here is more o an Ubuntu 20.04 installation. Here's a truncated error message (sorry, hit the gh limits):

```
   Compiling humility v0.7.2 (/home/rm/oxide/src/humility)
error: linking with `cc` failed: exit status: 1
  |
  = note: "cc" "-m64" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.0.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.1.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.10.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.11.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.12.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.13.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.14.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.15.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.2.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.3.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.4.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.5.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.6.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.7.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.8.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.humility.90a9a923-cgu.9.rcgu.o" "/home/rm/oxide/src/humility/target/release/deps/humility-720d3f1a49c94838.3u0183hbyawxcyog.rcgu.o" "-Wl,--as-needed" "-L" "/home/rm/oxide/src/humility/target/release/deps" "-L" "/home/rm/oxide/src/humility/target/release/build/capstone-sys-a103a19c39c53340/out" "-L" "/lib/x86_64-linux-gnu" "-L" "/home/rm/oxide/src/humility/target/release/build/hidapi-6b4cb0e445a9da99/out" "-L" "/lib/x86_64-linux-gnu" "-L" "/home/rm/oxide/src/humility/target/release/build/libusb1-sys-e4fd7d70907df78e/out" "-L" "/home/rm/oxide/src/humility/target/release/build/bzip2-sys-202871576d8801f5/out/lib" "-L" "/home/rm/.rustup/toolchains/1.58.0-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "/home/rm/oxide/src/humility/target/release/deps/libenv_logger-9fc5b3ac91d72c6e.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumantime-418c6f5073a18c14.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_validate-0b0b251e906341ba.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_trace-1485db5c9d25e376.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_test-c992c3728e524b9f.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_tasks-205d2dcafca7d0f0.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_stmsecure-4c9de33a7a2c7d1a.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_stackmargin-dacdc351e0a26327.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_spi-27c214d67143999b.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_spd-a507362e3decdc56.rlib" "/home/rm/oxide/src/humility/target/release/deps/libspd-4a2b034f6b7c80ca.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_spctrl-ac41aaf47d54cba6.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_sensors-02e2c6fba87bf4f6.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_ringbuf-c44c0503e3cab889.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_rendmp-6458e141e3d41c8d.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_rencm-d3bb029c80b1c008.rlib" "/home/rm/oxide/src/humility/target/release/deps/libserde_xml_rs-979aecec7a7154e0.rlib" "/home/rm/oxide/src/humility/target/release/deps/libxml-76d142ef27f78938.rlib" "/home/rm/oxide/src/humility/target/release/deps/libidt8a3xxxx-99674719a8b68151.rlib" "/home/rm/oxide/src/humility/target/release/deps/libitertools-3767bf843e821416.rlib" "/home/rm/oxide/src/humility/target/release/deps/libeither-3fbba4ae14cadab4.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_registers-45c46d34644aea72.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_readvar-bc1770e95059b11b.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_readmem-baf6b48be35b853f.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_qspi-784713ea6c81bfe4.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_probe-86718f3d4adcdf3b.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_pmbus-5eb66d4b7499257e.rlib" "/home/rm/oxide/src/humility/target/release/deps/libpmbus-0ceaf3b1bb5d92e1.rlib" "/home/rm/oxide/src/humility/target/release/deps/liblibm-9ff4e8496ec8ae68.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_openocd-465bc2914aebb8f5.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_monorail-b31a9508abb5802b.rlib" "/home/rm/oxide/src/humility/target/release/deps/libvsc7448_info-fad7de382f0bd3ff.rlib" "/home/rm/oxide/src/humility/target/release/deps/libvsc7448_types-81c671e4343bb0cd.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_map-b9961b8e517dd8d0.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_manifest-ab93147cf81828ae.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_lpc55gpio-9224ea20d65ae4e0.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_jefe-537771271b993436.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_itm-9a3eaca1699f45a9.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_i2c-08f76347a8fd470d.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_hiffy-2a39e738f9d226b9.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_hash-cd113e4fa5e4fe1d.rlib" "/home/rm/oxide/src/humility/target/release/deps/libsha2-b3f7fd5fa602893c.rlib" "/home/rm/oxide/src/humility/target/release/deps/libcpufeatures-6e3c852b071d46d0.rlib" "/home/rm/oxide/src/humility/target/release/deps/libdigest-85e4dcdcbc25bed3.rlib" "/home/rm/oxide/src/humility/target/release/deps/libblock_buffer-a380d58aefd12e43.rlib" "/home/rm/oxide/src/humility/target/release/deps/libcrypto_common-09f35b78ba3fe33f.rlib" "/home/rm/oxide/src/humility/target/release/deps/libgeneric_array-40d53e2b9909d265.rlib" "/home/rm/oxide/src/humility/target/release/deps/libtypenum-c18a35c735c947fc.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_gpio-6c1a8b9898c7a4e2.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_gdb-63fb2307268cbefc.rlib" "/home/rm/oxide/src/humility/target/release/deps/libctrlc-19ae07f06121ba5b.rlib" "/home/rm/oxide/src/humility/target/release/deps/libnix-435648b2b5ad7683.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_flash-4d3e028d0a56b231.rlib" "/home/rm/oxide/src/humility/target/release/deps/libsrec-bab177e6c9bb444b.rlib" "/home/rm/oxide/src/humility/target/release/deps/libtempfile-d56e20337343fdda.rlib" "/home/rm/oxide/src/humility/target/release/deps/libfastrand-7b156e81bdf19640.rlib" "/home/rm/oxide/src/humility/target/release/deps/libremove_dir_all-d2e21fa4e2946699.rlib" "/home/rm/oxide/src/humility/target/release/deps/libpath_slash-1d2fd126bb6ecb55.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_extract-e31ff5af23ce8576.rlib" "/home/rm/oxide/src/humility/target/release/deps/libhumility_cmd_exec-b144f84f49b8cd4a.rlib" "/home/rm/oxide/src/humility/target/release/deps/libsplitty-c [... cut ...]
          collect2: error: ld returned 1 exit status
          
  = help: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `humility` due to previous error
```

rmustacc, 2022-07-05:
Perhaps this is related to #174?

FawazTirmizi, 2022-08-09:
@rmustacc I am currently facing the same issue. Have you been able to resolve it?

FawazTirmizi, 2022-08-09:
Solved: There seems to be some bug with which version of libusb is selected. This also appears to happen on machines which upgraded from an older version of Ubuntu to 22.04. Uninstalling `libusb-0.1-4` fixed it for myself and multiple coworkers. Builds still succeed after reinstalling it.