oxidecomputer/humility #190: Flash Failing on Windows

tomioe, 2022-07-17:
Hi there.

I tried building and flashing Hubris to my STM32F3 Discovery Kit. I am doing this on Windows 10 `(version 21H2, OS Build 19044.1826)`.

```
c:\dev_ext\oxide\hubris>cargo xtask flash app\demo-stm32f4-discovery\app-f3.toml
    Finished dev [optimized + debuginfo] target(s) in 0.86s
     Running `target\debug\xtask.exe flash app\demo-stm32f4-discovery\app-f3.toml`
building crate task-jefe
[.. snip ..]
linking task 'jefe'
linking task 'rcc_driver'
linking task 'usart_driver'
linking task 'user_leds'
linking task 'ping'
linking task 'pong'
linking task 'hiffy'
linking task 'idle'
building crate demo-stm32f4-discovery
    Finished release [optimized + debuginfo] target(s) in 0.76s
target\demo-stm32f3-discovery\dist\kernel (unchanged)
flash = 0x08000000..0x08040000
ram   = 0x20000000..0x2000a000
Used:
  flash: 0x10000 (25%)
  ram:   0x8000 (80%)
humility: attaching with chip set to "STM32F301C6Tx"
humility: attached via ST-Link V2-1
humility flash failed: Flash loading failed Flash(NoSuitableNvm { start: 134250496, end: 134250528, description_source: External })
Error: humility failed
error: process didn't exit successfully: `target\debug\xtask.exe flash app\demo-stm32f4-discovery\app-f3.toml` (exit code: 1)
```

I am a bit confused about the `STM32F301C6Tx` line. My (every?) version of the STM32F3 Discovery Kit features the `STMF303VCT6`, with the ST-Link being handled by an `STM32F103C8T6`. I cannot even find any references to that particular MPN in any of the configs, so it's probably handled by a tool further down the chain.

For verification purposes, I'm including some extra information here

```
c:\dev\rust_embedded>humility --version
humility 0.7.3
c:\dev\rust_embedded>cargo --version
cargo 1.62.0 (a748cf5a3 2022-06-08)
c:\dev\rust_embedded>type openocd_run.bat
cd c:\OpenOCD-w64
openocd.exe -s ..\scripts\ -f ..\scripts\interface\stlink.cfg -f ..\scripts\target\stm32f3x_stlink.cfg
cd c:\dev\rust_embedded
c:\dev\rust_embedded>openocd_run.bat

c:\dev\rust_embedded>cd c:\OpenOCD-w64

c:\OpenOCD-w64>openocd.exe -s ..\scripts\ -f ..\scripts\interface\stlink.cfg -f ..\scripts\target\stm32f3x_stlink.cfg
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev (2020-10-13-17:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
WARNING: target/stm32f3x_stlink.cfg is deprecated, please switch to target/stm32f3x.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J27M15 (API v2) VID:PID 0483:374B
Info : Target voltage: 2.907884
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f3x.cpu on 3333
Info : Listening on port 3333 for gdb connections
shutdown command invoked
```

As shown, I can connect to the board just fine with OpenOCD on its own.

Has anyone had the same experience? Any suggestions as to what I need to change to get it running?

labbott, 2022-07-18:
This is probably related to https://github.com/oxidecomputer/humility/pull/115 . We might not be passing the correct chip version for the F3 discovery board you have. It might be easiest to get a copy of `probe-rs` and figure out the correct chip version to pass. We're also supposed to fall back to OpenOCD if that doesn't work so it would be worthwhile to investigate why that didn't happen either.

tomioe, 2022-07-18:
> It might be easiest to get a copy of probe-rs and figure out the correct chip version to pass

This is the output I get:

```
c:\dev_ext\probe-rs\target\release>probe-rs-cli.exe list
The following devices were found:
[0]: STLink V2-1 (VID: 0483, PID: 374b, Serial: 066FFF484951717867055217, StLink

c:\dev_ext\probe-rs\target\release>probe-rs-cli.exe chip info STM32F303VCT6
STM32F303VCT6
Cores (1):
    - main (Armv7em)
RAM: 0x10000000..0x10002000 (8.00 KiB)
NVM: 0x08000000..0x08040000 (256.00 KiB)
```

I also ran `probe-rs-cli.exe info`, please find the [the output here.](https://gist.github.com/tomioe/f5f345f3390886d578244197f3c71933).

My attempt to manually flash the kernel was "unsuccessful", but I'm probably not using the right command:

```
c:\dev_ext\probe-rs\target\release>probe-rs-cli.exe run c:\dev_ext\oxide\hubris\target\demo-stm32f3-discovery\dist\kernel
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: Unable to load specification for chip
    1: The connected chip could not automatically be determined.
```

P.s. I wrote the wrong chip in my original comment, it's of course the STM32F303VCT6.

tomioe, 2022-07-22:
Church

closed 2022-07-22 by tomioe