oxidecomputer/humility #191: Reads to FPGA1 on QSFP-IO board after reading FPGA0 requires a repeat

ericaasen, 2022-07-18:
When reading a register from FPGA1 after reading a register from FPGA0, it seems to require two read accesses to get the correct value.  Here I am reading the PHY status register from FPGA1 after I read the same register from FPGA0.  I expect 0x0 from FPGA0 and 0x77 from FPGA1:

```
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=0,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x0
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x0
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0xb
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x77
```

example of writes taking effect immediately (register value on both FPGAs = 0xDE before being written to):
```
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_write_reg -adevice_index=0,addr=0x5,value=0x01,op=Write
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_write_reg() = ()
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_write_reg -adevice_index=1,addr=0x5,value=0x02,op=Write
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_write_reg() = ()
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=0,addr=0x5
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x0
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=0,addr=0x5
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x1
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0x5
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x0
eric@niles ~ (sidecar-sp) $ pfexec $HUMILITY hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -adevice_index=1,addr=0x5
humility: attached to 0483:374f:002A001C4D46500F20373033 via ST-Link V3
Fpga.user_design_read_reg() = 0x2
```

mkeeter, 2022-07-25:
This _may_ not be a Hubris / Humility issue; we've seen weirdness where you have to read the FPGA ID twice before it returns correct data:
https://github.com/oxidecomputer/hardware-qsfp-x32/issues/36

cc @Aaron-Hartwig

mkeeter, 2022-07-26:
Fixed in Hubris#606 above

closed 2022-07-26 by mkeeter