oxidecomputer/humility #268: Bad ringbuf decoding

mkeeter, 2022-11-29:
See attached dump!

```console
➜  ~ humility -d hubris.core.bad-ringbuf ringbuf task_monorail_server::bsp::__RINGBUF
humility: attached to dump
humility: ring buffer task_monorail_server::bsp::__RINGBUF in monorail:
humility: ringbuf dump failed: loading value of type struct UnsafeCell<ringbuf::Ringbuf<task_monorail_server::bsp::Trace, 16>> at address 0x0
```
[hubris.core.bad-ringbuf.zip](https://github.com/oxidecomputer/humility/files/10114930/hubris.core.bad-ringbuf.zip)

mkeeter, 2022-11-29:
Running `humility readvar` shows the following more detailed error:
```
humility readvar failed: loading value of type struct StaticCell<ringbuf::Ringbuf<task_monorail_server::bsp::Trace, 16>> at address 0x0

Caused by:
    0: loading value of type struct UnsafeCell<ringbuf::Ringbuf<task_monorail_server::bsp::Trace, 16>> at address 0x0
    1: loading value of type struct Ringbuf<task_monorail_server::bsp::Trace, 16> at address 0x0
    2: loading value of type array at address 0x0
    3: loading value of type struct RingbufEntry<task_monorail_server::bsp::Trace> at address 0x0
    4: loading value of type enum Trace at address 0x0
    5: loading value of type struct AnegCheckFailed at address 0x0
    6: loading value of type enum VscError at address 0x0
    7: unknown variant: 0x32
```

This smells a lot like #263 (meaning address `0x0` is a red herring)

Aaron-Hartwig, 2022-12-05:
This happened to me as well.
```
humility -d hubris.core.bad-ringbuf-2 ringbuf  drv_transceivers_server::__RINGBUF
humility: attached to dump
humility: ring buffer drv_transceivers_server::__RINGBUF in transceivers:
humility: ringbuf dump failed: loading value of type struct UnsafeCell<ringbuf::Ringbuf<drv_transceivers_server::Trace, 16>> at address 0x0
```
[hubris.core.bad-ringbuf-2.zip](https://github.com/oxidecomputer/humility/files/10155922/hubris.core.bad-ringbuf-2.zip)

Aaron-Hartwig, 2022-12-05:
It seems like `ringbuf` can no longer be trusted at all as it is not decoding entries correctly. For example, in `transceivers`:
```
pfexec humility -t sidecar-b -a /home/aaron/build-sidecar-b.zip ringbuf transceivers
humility: WARNING: archive on command-line overriding archive in environment file
humility: attached to 0483:3754:002600184D4B500E20373831 via ST-Link V3
humility: ring buffer drv_transceivers_server::__RINGBUF in transceivers:
 NDX LINE      GEN    COUNT PAYLOAD
   0  326        1        1 FrontIOReady(true)
   1  359        1        1 LEDInit
   2   98        1        1 LEDInitComplete
   3  306        1        1 LEDErrorSummary(FullErrorSummary { overtemp_left: true, overtemp_right: false, system_led_err: NoError, open_circuit: 0xffffffff, short_circuit: 0x809f679, invalid: 0x20 })
```
Note that index 3 has an `LEDErrorSummary` message from line 306, but that should actually be a `ModulePresenceUpdate` https://github.com/oxidecomputer/hubris/blob/master/drv/transceivers-server/src/main.rs#L306

closed 2022-12-05 by mkeeter