oxidecomputer/humility #347: panic at the probe-rs

jclulow, 2023-04-04:
We have had some panics in probe-rs in `humility` during programming this week.  I have attempted to characterise them here.

Some of these panics have been transient for the Gimlet in question and the Gimlet has gone on to have a successful programming after at least one retry.  Some of the Gimlets have not yet cleared programming at least in part on account of this stuff happening.

Note that unfortunately some of these stacks are not complete: due to oxidecomputer/facade#53 and the fact that
 the `humility` process appears to actually abort at least some of the time, and because our log records are written over a UNIX socket to another process, I believe the facade UI is getting knee-capped by this failing child process before it can write it all out.  I will fix that, but I suspect the original panic _message_ is probably the most useful thing to get started here anyway.

### Unique Panic Messages:

- ``thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', /home/jclulow/.cargo/git/checkouts/probe-rs-c1a8b38a079f9523/ca13d23/probe-rs/src/probe/cmsisdap/mod.rs:599:12``
- ``thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', /home/jclulow/.cargo/git/checkouts/probe-rs-c1a8b38a079f9523/d731eaa/probe-rs/src/probe/cmsisdap/mod.rs:599:16``
- ``thread 'main' panicked at 'index out of bounds: the len is 2 but the index is 2', /home/jclulow/.cargo/git/checkouts/probe-rs-c1a8b38a079f9523/ca13d23/probe-rs/src/probe/cmsisdap/commands/transfer/mod.rs:186:24``
- ``thread 'main' panicked at 'index out of bounds: the len is 2 but the index is 2', /home/jclulow/.cargo/git/checkouts/probe-rs-c1a8b38a079f9523/d731eaa/probe-rs/src/probe/cmsisdap/commands/transfer/mod.rs:186:24``
- ``thread 'main' panicked at 'index out of bounds: the len is 3 but the index is 3', /home/jclulow/.cargo/git/checkouts/probe-rs-c1a8b38a079f9523/d731eaa/probe-rs/src/probe/cmsisdap/commands/transfer/mod.rs:186:24``
- ``thread panicked while panicking. aborting.``

Because GitHub is a ridiculous farce, I will have to split up the more complete logs into comments in the issue.

jclulow, 2023-04-04:
### Gimlet Rev C BRM44220003 (activity `01GW0EV7ZNDDGW6DZJP14KDDK2`)

```
WARNING: last command (pid 1897) still outstanding:
exec: {"args":["/opt/oxide/dvt/scripts/rot.sh","gimlet"],"envs":{"LOOM_GIMLET_IGNITION_SERIAL":"FT8BF844","LOOM_GIMLET_MCULINK_PID":"0143","LOOM_GIMLET_UART_VID":"0403","POWER_KEY":"4tDCoSASt7c9EvE1MCpH83iaghWzHZHgNLQA0JztUfUDFCC80q5W6Qpc3BobfSPu","LOOM_GIMLET_STLINK_PID":"3754","LOOM_GIMLET_UART_PID":"6010","LOOM_GIMLET_STLINK_VID":"0483","LOOM_GIMLET_MCULINK_SERIAL":"JENXLKBA0YBLT","LOOM_GIMLET_MCULINK_VID":"1fc9","LOOM_GIMLET_IGNITION_PID":"6014","LOOM_GIMLET_UART_SERIAL":"FT5X1WQ2","LOOM_GIMLET_IGNITION_VID":"0403","LOOM_GIMLET_STLINK_SERIAL":"0039001D4741500920383733"}}
    |O|  * turning power on...
    |O| power is now on
    |O|  * setting debug mode...
    |O| Looks like a plausible debug mailbox
    |O| Reset chip successfully!
    |O| entering debug
    |O|  * checking ROT stage0...
    |E| humility: attaching with chip set to "LPC55S69JBD100"
    |E| [2023-03-20T21:30:19Z WARN  probe_rs::config::target] Using custom sequence for LPC55S69
    |E| humility: attached to 1fc9:0143:JENXLKBA0YBLT via CMSIS-DAP
    |E| humility flash failed: image IDs match, but flash contents do not match archive contents: expected structure RegionDesc not found
    |O|  * programming ROT stage0...
    |E| humility: attaching with chip set to "LPC55S69JBD100"
    |E| [2023-03-20T21:30:20Z WARN  probe_rs::config::target] Using custom sequence for LPC55S69
    |E| humility: attached to 1fc9:0143:JENXLKBA0YBLT via CMSIS-DAP
    |E| humility: image IDs match, but flash contents do not match archive contents: expected structure RegionDesc not found; reflashing
    |E| humility: flashing done
    |O|  * ROT stage0 ok!
    |O|  * checking ROT image...
    |E| humility: attaching with chip set to "LPC55S69JBD100"
    |E| [2023-03-20T21:30:22Z WARN  probe_rs::config::target] Using custom sequence for LPC55S69
    |E| humility: attached to 1fc9:0143:JENXLKBA0YBLT via CMSIS-DAP
    |E| thread 'main' panicked at 'index out of bounds: the len is 2 but the index is 2', /home/jclulow/.cargo/git/checkouts/probe-rs-c1a8b38a079f9523/ca13d23/probe-rs/src/probe/cmsisdap/commands/transfer/mod.rs:186:24
    |E| note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
    |E| thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', /home/jclulow/.cargo/git/checkouts/probe-rs-c1a8b38a079f9523/ca13d23/probe-rs/src/probe/cmsisdap/mod.rs:599:12
    |E| stack backtrace:
    |E|    0:          0x2726d1c - std::backtrace_rs::backtrace::libunwind::trace::ha40da32d92ac47c0
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93:5
    |E|    1:          0x2726d1c - std::backtrace_rs::backtrace::trace_unsynchronized::h290da7823e73f239
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
    |E|    2:          0x2726d1c - std::sys_common::backtrace::_print_fmt::h4eb3e237051e106f
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/sys_common/backtrace.rs:66:5
    |E|    3:          0x2726d1c - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h1b08402c080c8867
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/sys_common/backtrace.rs:45:22
    |E|    4:          0x277d6fa - core::fmt::write::he355916a414285f6
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/core/src/fmt/mod.rs:1202:17
    |E|    5:          0x27188e4 - std::io::Write::write_fmt::ha5a54ba24b8a9d8c
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/io/mod.rs:1679:15
    |E|    6:          0x27297cb - std::sys_common::backtrace::_print::h1976e5c6b44ae641
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/sys_common/backtrace.rs:48:5
    |E|    7:          0x27297cb - std::sys_common::backtrace::print::hbd5ddbc7ffa1214f
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/sys_common/backtrace.rs:35:9
    |E|    8:          0x27297cb - std::panicking::default_hook::{{closure}}::h446cd217aeac857c
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/panicking.rs:295:22
    |E|    9:          0x272941c - std::panicking::default_hook::h8f5294e06706b5c7
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/panicking.rs:314:9
    |E|   10:          0x2729fb1 - std::panicking::rust_panic_with_hook::h7d1849cd4f667ca2
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/panicking.rs:698:17
    |E|   11:          0x2729e72 - std::panicking::begin_panic_handler::{{closure}}::h8a45f063f3c93d78
    |E|                                at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/panicking.rs:586:13
    |E|   12:          0x27271c0 - std::sys_common::backtrace::__rust_end_short_backtrace::hf8c1e6acdf9f488d
[... cut ...]
    |E|   10:          0x2729e20 - std::panicking::rust_panic_with_hook::he9bc2549af37a109
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:688:13
    |E|   11:          0x2729b63 - std::panicking::begin_panic_handler::{{closure}}::h6df748414270e38e
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:577:13
    |E|   12:          0x2726ec0 - std::sys_common::backtrace::__rust_end_short_backtrace::h2c4715300e4fba8f
```

jclulow, 2023-04-04:
### Gimlet Rev C BRM44220003 (activity `01GX4NCZGJZKN90FTCJNVZRAVF`)

```
WARNING: last command (pid 1057) still outstanding:
exec: {"args":["/opt/oxide/dvt/scripts/rot.sh","gimlet"],"envs":{"LOOM_GIMLET_MCULINK_PID":"0143","LOOM_GIMLET_UART_VID":"0403","POWER_KEY":"v3YVyWgiZHWP8XTVaqBtwczkLgZ1PxlFgsyfGf6J3Cykt11s8xChzBvOz86LPspg","LOOM_GIMLET_IGNITION_SERIAL":"FT8BF844","LOOM_GIMLET_IGNITION_VID":"0403","LOOM_GIMLET_STLINK_SERIAL":"0039001D4741500920383733","LOOM_GIMLET_IGNITION_PID":"6014","LOOM_GIMLET_MCULINK_VID":"1fc9","LOOM_GIMLET_MCULINK_SERIAL":"JENXLKBA0YBLT","LOOM_GIMLET_STLINK_VID":"0483","LOOM_GIMLET_STLINK_PID":"3754","LOOM_GIMLET_UART_PID":"6010","LOOM_GIMLET_UART_SERIAL":"FT5X1WQ2"}}
    |O|  * image "/data/local/images/gimlet/c/rot0lab/build-gimlet-rot-c-lab.zip" is:
    |O|      git rev => 715a2636ac45ad53c5dae8b767c793a27b23ef0e
    |O|     image id => [66, ce, fc, ee, 1e, b4, 3f, 23]
    |O| 
    |O|  * image "/data/local/images/gimlet/c/rot/build-gimlet-rot-c.zip" is:
    |O|      git rev => 715a2636ac45ad53c5dae8b767c793a27b23ef0e
    |O|     image id => [52, 2b, a5, 50, 14, d0, d, e3]
    |O| 
    |O|  * turning power on...
    |O| power is now on
    |O|  * setting debug mode...
    |O| Looks like a plausible debug mailbox
    |O| Reset chip successfully!
    |O| entering debug
    |O|  * checking ROT stage0...
    |E| humility: attaching with chip set to "LPC55S69JBD100"
    |E| [2023-04-03T22:58:06Z WARN  probe_rs::config::target] Using custom sequence for LPC55S69
    |E| humility: attached to 1fc9:0143:JENXLKBA0YBLT via CMSIS-DAP
    |E| humility flash failed: flash/archive mismatch: image ID in archive ([66, ce, fc, ee, 1e, b4, 3f, 23]) does not equal ID at 0x75d0 ([ba, 6f, 17, 72, aa, 67, f0, 6])
    |O|  * programming ROT stage0...
    |E| humility: attaching with chip set to "LPC55S69JBD100"
    |E| [2023-04-03T22:58:06Z WARN  probe_rs::config::target] Using custom sequence for LPC55S69
    |E| humility: attached to 1fc9:0143:JENXLKBA0YBLT via CMSIS-DAP
    |E| thread 'main' panicked at 'index out of bounds: the len is 3 but the index is 3', /home/jclulow/.cargo/git/checkouts/probe-rs-c1a8b38a079f9523/d731eaa/probe-rs/src/probe/cmsisdap/commands/transfer/mod.rs:186:24
    |E| note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
    |E| thread 'main' panicked at 'index out of bounds: the len is 2 but the index is 2', /home/jclulow/.cargo/git/checkouts/probe-rs-c1a8b38a079f9523/d731eaa/probe-rs/src/probe/cmsisdap/commands/transfer/mod.rs:186:24
    |E| stack backtrace:
    |E|    0:          0x2726a16 - std::backtrace_rs::backtrace::libunwind::trace::h2c4f45e9662b7c2e
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93:5
    |E|    1:          0x2726a16 - std::backtrace_rs::backtrace::trace_unsynchronized::h5b2b95ab06f58282
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
    |E|    2:          0x2726a16 - std::sys_common::backtrace::_print_fmt::h47fd4a9c37b51847
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/sys_common/backtrace.rs:65:5
    |E|    3:          0x2726a16 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h04db6524de7f1732
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/sys_common/backtrace.rs:44:22
    |E|    4:          0x278487a - core::fmt::write::h3e5c475834498b24
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/core/src/fmt/mod.rs:1213:17
    |E|    5:          0x2717e54 - std::io::Write::write_fmt::h2b69f22ea2882b36
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/io/mod.rs:1682:15
    |E|    6:          0x27267f0 - std::sys_common::backtrace::_print::h72c971df88deaae1
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/sys_common/backtrace.rs:47:5
    |E|    7:          0x27267f0 - std::sys_common::backtrace::print::h19ad093482386556
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/sys_common/backtrace.rs:34:9
    |E|    8:          0x27294f3 - std::panicking::default_hook::{{closure}}::h9ec99b6caf7b7c99
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:267:22
    |E|    9:          0x272918d - std::panicking::default_hook::hd7515996e070a27e
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:286:9
    |E|   10:          0x2729e20 - std::panicking::rust_panic_with_hook::he9bc2549af37a109
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:688:13
    |E|   11:          0x2729ba6 - std::panicking::begin_panic_handler::{{closure}}::h6df748414270e38e
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:579:13
    |E|   12:          0x2726ec0 - std::sys_common::backtrace::__rust_end_short_backtrace::h2c4715300e4fba8f
[... cut ...]
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:267:22
    |E|    9:          0x272918d - std::panicking::default_hook::hd7515996e070a27e
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:286:9
    |E|   10:          0x2729e20 - std::panicking::rust_panic_with_hook::he9bc2549af37a109
    |E|                                at /rustc/2c8cc343237b8f7d5a3c3703e3a87f2eb2c54a74/library/std/src/panicking.rs:688:13
```

bcantrill, 2023-04-06:
To try to make this failure mode a little cleaner, we pulled in the fix for https://github.com/probe-rs/probe-rs/issues/922, in which a similar issue hads been encountered.  Turning on verbose output (with `humility -v`) and then capturing the failure:

```
[2023-04-04T23:29:36Z DEBUG probe_rs::architecture::arm::ap] Writing register TAR, value=TAR { address: e000edf4 }
[2023-04-04T23:29:36Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Write(port=AccessPort, addr=4, data=0xe000edf4
[2023-04-04T23:29:36Z DEBUG probe_rs::architecture::arm::ap] Writing register DRW, value=DRW { data: e }
[2023-04-04T23:29:36Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Write(port=AccessPort, addr=12, data=0x0000000e
[2023-04-04T23:29:36Z DEBUG probe_rs::architecture::arm::ap] Writing register TAR, value=TAR { address: e000edf0 }
[2023-04-04T23:29:36Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Write(port=AccessPort, addr=4, data=0xe000edf0
[2023-04-04T23:29:36Z DEBUG probe_rs::architecture::arm::ap] Reading register DRW
[2023-04-04T23:29:36Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Read(port=AccessPort, addr=12)
[2023-04-04T23:29:36Z DEBUG probe_rs::probe::cmsisdap] 4 items in batch
[2023-04-04T23:29:36Z DEBUG probe_rs::probe::cmsisdap] Attempting batch of 4 items
[2023-04-04T23:29:36Z TRACE probe_rs::probe::cmsisdap::commands] Transmit buffer: [00, 05, 00, 04, 05, F4, ED, 00, E0, 0D, 0E, 00, 00, 00, 05, F0, ED, 00, E0, 0F, 00]...
[2023-04-04T23:29:36Z DEBUG probe_rs::architecture::arm::ap] Writing register CSW, value=CSW { DbgSwEnable: 0, HNONSEC: 0, PROT: 2, CACHE: b, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 0, AddrInc: Single, _RES1: 0, SIZE: U32 }
[2023-04-04T23:29:36Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Write(port=AccessPort, addr=0, data=0x2b000012
[2023-04-04T23:29:36Z DEBUG probe_rs::architecture::arm::ap] Writing register TAR, value=TAR { address: e000edf0 }
[2023-04-04T23:29:36Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Write(port=AccessPort, addr=4, data=0xe000edf0
[2023-04-04T23:29:36Z DEBUG probe_rs::architecture::arm::ap] Writing register DRW, value=DRW { data: a05f000d }
[2023-04-04T23:29:36Z DEBUG probe_rs::probe::cmsisdap] Adding command to batch: Write(port=AccessPort, addr=12, data=0xa05f000d
[2023-04-04T23:29:36Z DEBUG probe_rs::probe::cmsisdap] 3 items in batch
[2023-04-04T23:29:36Z DEBUG probe_rs::probe::cmsisdap] Attempting batch of 3 items
[2023-04-04T23:29:36Z TRACE probe_rs::probe::cmsisdap::commands] Transmit buffer: [00, 05, 00, 03, 01, 12, 00, 00, 2B, 05, F0, ED, 00, E0, 0D, 0D, 00, 5F, A0, 00]...
[2023-04-04T23:29:36Z TRACE probe_rs::probe::cmsisdap::commands] Receive buffer: [05, 04, 01, 03 ...
[2023-04-04T23:29:36Z ERROR probe_rs::probe::cmsisdap::commands::transfer] Transfer count larger than requested number of transfers
```	

This is revealing:  we do not see a `Receive buffer` trace message after the
first `Transmit buffer` in the snippet -- and indeed, what we *do* see is
an an attempt to write CSW.  This is bad news:  we have bombed out with
the probe in an inconsistent state (namely, with data to be pulled from the
wire in response to work that it is abandoning), and re-entering to write
the CSW is leading us to the error message with respect to transfer count.
These leaves two questions:  why are we failing and why are we writing CSW
after we have failed?  For the latter,
logging a stack backtrace on this (unexpected)
CSW write reveals that it's coming from Humility trying to run the core
after failing to flash it.  This is really not a good idea:  if flashing
fails -- the core is in an entirely undefined state -- and it's a very
bad idea to do this in such a way that the original error is effectively
dropped!  Fixing this issue is straightforward:

```diff
diff --git a/cmd/flash/src/lib.rs b/cmd/flash/src/lib.rs
index c5c6900..b9fc26b 100644
--- a/cmd/flash/src/lib.rs
+++ b/cmd/flash/src/lib.rs
@@ -332,12 +332,12 @@ fn flashcmd(context: &mut humility::ExecutionContext) -> Result<()> {
     let ihex_path = ihex.path();
 
     //
-    // Load the flash image, and reset the part if that works.
+    // Load the flash image.  If that fails, we're in a world of hurt:  we
+    // really don't want to run the core for fear of masking of initial
+    // error.  (It will hopefully be pretty clear to the user that a
+    // half-flashed part is going to be in an ill-defined state!)
     //
-    if let Err(err) = core.load(ihex_path) {
-        core.run()?;
-        return Err(err);
-    }
+    core.load(ihex_path)?;
 
     //
     // On Gimlet Rev B, the BOOT0 pin is unstrapped -- and during a flash,
```

Fixing this issue gives us a much cleaner failure and shines light on our remaining question (namely, why is this failing at all):

```
humility flash failed: Flash loading failed Flash(PageWrite { page_address: 183296, source: ArchitectureSpecific(RegisterReadError { address: 12, name: "DRW", source: ProbeSpecific(Send { command_id: Transfer, source: Timeout }) }) })
```

Knowing that we have failed due to a timeout allows us to return to the CMSIS-DAP code, where the USB timeout value code is 100 milliseconds.  This is quite a bit more aggressive than either ST-Link (1000 milliseconds) or Espressif (5000 milliseconds).  Modifying this value and then recording the latency over time on the machine in question at our manufacturing facility reveals that 100 millisec [... cut ...]
On the one hand, there are some open questions on why we are seeing this on this machine in our facility and not the systems in Emeryville -- but on the other, the (arbitrary) timeout is indisputably too short.

Indeed, this has been seen before in terms of https://github.com/probe-rs/probe-rs/issues/1423.  While that fix is narrower than we would like (it only addresses one of the 100 millisecond timeouts) and not necessarily as clean (it merely replaces one hard-coded constant with another, without any additional comment or explanation), it also results in the least divergence from upstream -- and cherry-picking the commit resolves our issue here.

commit 0aad9828 (2023-04-06): panic at the probe-rs

- Fixes #347
- Fixes #262
- Adds an option to "humility spd" to emit SPD data to a binary file
- Cleans up the help message for Humility to both better assist and to
  point to "humility doc"
PR #350 (2023-04-06): panic at the probe-rs
- Fixes #347
- Fixes #262
- Adds an option to "humility spd" to emit SPD data to a binary file
- Cleans up the help message for Humility to both better assist and to point to "humility doc"
commit 6c2704fe (2023-04-06): panic at the probe-rs (#350)

- Fixes #347
- Fixes #262
- Adds an option to "humility spd" to emit SPD data to a binary file
- Cleans up the help message for Humility to both better assist and to
  point to "humility doc"
closed 2023-04-06 by bcantrill