oxidecomputer/humility #390: "humility rendmp --phase-check" on VDD_MEM_ABCD can result in locked I2C bus

bcantrill, 2023-05-27:
On some machines, running `humility rendmp --phase-check` results in a very locked I2C bus.  To the user, this will look like this:

```console
$ humility rendmp --phase-check --device 0x5a
humility: attached via ST-Link V3
Phase check for RAA229618 at 0x5a
PHASE |   VOUT   |   IOUT   |   TEMP   | RAIL
------|----------|----------|----------|----------
humility: restoring default configuration...
humility: WARNING: error restoring default cfg when recovering from error: failed to restore default cfg

Caused by:
    BusReset
humility rendmp failed: failed to READ_VOUT: ControllerBusy
```

At this point, the bus is borked, e.g.:

```console
$ humility i2c -s -b mid
humility: attached via ST-Link V3

Device scan on controller I2C3:

    R = Reserved   - = No device   \o/ = Device found   X = Timed out

ADDR     0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf
0x00       R   R   R   R   R   R   R   R Err Err Err Err Err Err Err Err
0x10     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x20     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x30     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x40     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x50     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x60     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x70     Err Err Err Err Err Err Err Err Err Err Err Err   R   R   R   R

Error summary:

  COUNT ERROR
    116 BusReset

```

Here is logic analyzer output of the mid bus when it locks:

[image]

From this point on, SDA is held low and SCL is high.  Of note, resetting the SP does *not* result in the bus being let go -- and indeed, the SCL wiggles that are a part of I2C initialization do not result in SDA being released, e.g.:

[image]

A few things to note about this:

- This always happens immediately after 0 has been written to `ON_OFF_CONFIG`, enabling the rail in PWM mode
- There is always a ~150  μsec delay between the final stop condition that is the write to `ON_OFF_CONFIG` and SDA being pulled low
- On Gimlet, this has only been seen on `VDD_MEM_ABCD` -- and not on `VDD_VCORE` (which has different phases but otherwise shares the same controller)
- It happens on about 80% of the Gimlets that I tried (both C and B) -- but there *are* Gimlets that are able to perform `--phase-check` cleanly on all rails
- The only thing that appears to clear the condition is a return to A3 (that is, powering off the RAA229618).

bcantrill, 2023-06-01:
@ericaasen asked for a full [Salaea capture of the mid bus](https://github.com/oxidecomputer/humility/files/11621126/midbus.zip).  That capture shows a power on, followed by a `humility rendmp  --phase-check --device 0x5b` (which works) followed by a `humility rendmp --phase-check --device 0x5a` (which hangs). I also did the experiment of tri-stating the I2C pins via `humility gpio`; it did not unhang the bus.  (Which is not surprising given that resetting the SP *also* did not unhang the bus).

bcantrill, 2023-06-07:
@cbiffle had a hypothesis observation:

> So based on the ground-offset observed in the I2C bus hang, I believe the culprit to be on the A-side of U546 (TCA9803), whereas J228 (the connector used to make the observations) and U350 (the RAA229618 being spoken to) are on the B-side of U546. U350 was observed as being able to pull the data line to ~0V relative to J228's ground pin, which tracks.
>
> However, the TCA9803 itself doesn't introduce a ground offset under normal operation. But when the bus hang, we saw a ~220 mV offset, and even when the SP was pulling lines low we saw the same offset (observed at J228). The SP is the only device on the nets between U546 and another translator, U515, an LSF0108. Tristating the SP's pins did not fix the bus being held low, and did not affect the ground offset.
>
> The way the LSF0108 works, it does introduce a ground offset, as a result of (complicated analysis expanded upon in that 8-episode video series). This has me concerned that the device that takes the bus out may be on the far side of the LSF0108, in the V2P5 mid-bus domain. The only devices on that segment, as far as I can tell, are the ABCD DIMM SPDs.
>
> The operation being performed when the bus hung was a rail test on U350 (RAA229618), which...happens to produce the power for the ABCD DIMMs. The A0 power, mind, not the A2 SPD power. But the combination seems suspicious.
> 
> Neither the TCA9803 nor the LSF0108 seem capable of latchup, afaict, so I'm now wondering if this might be a DIMM SPD acting up. At some point middle-of-next-week when both bcantrill and I are in EMY physically, we're going to re-run the test with the ABCD DIMMs removed, which will either quickly rule this out, or suggest a culprit.
>
> That being said, if the answer is "the DIMMs act weird when you glitch-attack them," that's good to know but not relevant for our other I2C bus hangs.

To explore this, we pulled the DIMMs on a Gimlet that has very reliably reproduced this -- and sure enough:

[image]

So at the very least, this is DIMM related. We are continuing to investigate.

ericaasen, 2023-06-07:
per the Micron datasheet, the register device on the DIMM connects to I2C:

[image]

https://media-www.micron.com/-/media/client/global/documents/products/data-sheet/modules/rdimm/ddr4/asf36c8gx72pz.pdf?rev=52efd03dbf8648a5b59fdc7d1cd4bb31

ericaasen, 2023-06-07:
Per the [RCD spec](https://drive.google.com/file/d/1nayVXGL3vFRxnqAIbGgbAbOknsA_1zjW/view), VDD for the register is 1.2V, which is VDD_ABCD on Gimlet, but the I2C interface is powered by VDDSPD, which is always on in our system.  My suspicion is there is enough internal isolation that if the 1.2V rail is not present on a DIMM at all, the RCD will leave the I2C bus pins in tristate as it should, but depending on the specific RCD, the voltage that this test ends up producing powers up enough circuitry in the RCD to have it cause trouble with the I2C pins, leading to our bus locked state.

bcantrill, 2023-06-07:
We have found that on *some* DIMMs (3 of the 16 on the Rev C Gimlet that we have been using), this *always* happens when the DIMM is in any slot on the ABCD side. Our conclusion -- per @ericaasen's observation above -- is that the RCD is in an undefined state at the voltage levels that we're using for the phase tests (400 mV), and that on some DIMMs this results in the I2C bus that they are (covertly) on to be pulled down.  The fix for this is to refuse to perform the `--phase-check` if DIMMs are found to be populated (with an added `--force-phase-check` flag to force it).

It is a bit of a mystery *why* this only happens to some DIMMs -- though as @ericaasen and @cbiffle have pointed out, this could be due to many things like process variation.  Finally, note that the observed rate of DIMM lockup (3/16th -- 18.75%) and the requirements to have non-lockup (namely, 8 non-failing DIMMs on the ABCD side) would result in a rate of Gimlet non-lockup (0.8125<sup>8</sup> = 0.1899) that matches initial rough observations (~0.2).  While this may be somewhat coincidental, it feels at least directionally consistent!

PR #391 (2023-06-08): "humility power", "humility rendmp --phase-check" tweaks
Fixes #390. Draft comment:

```
This fixes #390 and makes some other small tweaks to "humility power"
(taking a rail as an argument) and "humility rendmp --phase-check"
(taking a phase as an argument + cleaning up output a tad).
```
commit 40858d24 (2023-06-08): "humility power", "humility rendmp --phase-check" tweaks (#391)

This fixes #390 and makes some other small tweaks to "humility power"
(taking a rail as an argument) and "humility rendmp --phase-check"
(taking a phase as an argument + cleaning up output a tad).
closed 2023-06-08 by bcantrill