oxidecomputer/humility #408: humility flash should verify auxflash contents independent of whether primary ROM was intact

jclulow, 2023-07-27:
For archives that contain not just a primary hubris image but also an auxiliary flash image, `humility flash` will write the auxflash image into the next slot iff it is not already present.  It appears that we presently do not _verify_ that the auxflash image is present and complete iff the primary hubris image is completely intact in the ROM at the time of the check.

During manufacturing, when programming the SP, we perform essentially this sequence of commands:

```bash
if ! humility flash --check; then
        printf 'contents incorrect; programming...\n'
        if ! humility flash --verify; then
                printf 'programming failed!\n'
                exit 1
        fi
fi
printf 'ok\n'
```

On a programming station, we were having a reproducible issue that occurred each time we tried to erase and write an auxiliary flash slot on a **Sidecar/B**.  The precise nature of the issue is not germane; the upshot is that the hubris image was written, but the auxflash write was left incomplete.

When the issue was corrected and we went to retry programming, it appears that `humility flash --check` confirmed that the hubris image in the device had a valid checksum and then _did not check the auxflash_.  When we went to power on the system, the system was unable to start because the FPGA image was not available.

A manual `humility flash -F` corrected the issue: it reported that it did not _need_ to write the hubris image again, but did anyway; it then did the auxflash update correctly.

With my apologies if this has already been fixed in a later versoin, this behaviour was seen with:

```
$ humility --version
humility 0.10.24
```

mkeeter, 2023-07-27:
I opened a PR to fix this.  One quick aside:

```
if ! humility flash --check; then
        printf 'contents incorrect; programming...\n'
        if ! humility flash --verify; then
                printf 'programming failed!\n'
                exit 1
        fi
fi
printf 'ok\n'
```
doesn't seem quite right. `humility flash --verify` checks the image _before_ flashing, so it duplicates the work done by `humility flash --check`, and we don't have any post-flashing checks.

It would be more reasonable to do
```
if ! humility flash --verify; then
        printf 'programming failed!\n'
        exit 1
fi
if ! humility flash --check; then
        printf 'contents incorrect after programming!\n'
        exit 1
fi
printf 'ok\n'

jclulow, 2023-07-27:
It has been a while since I looked at this, but I believe the reason I did it this way is that `humility flash --verify` _fails_ iff there was no work to do; i.e., if it verified that the ROM was already in place, it exits non-zero.  That's not what we want at all; rather what we really want is:

```
if ! humility flash --idempotent; then
        printf 'problems!\n'
        exit 1
fi
```

We are indeed duplicating the check, as you note, in the case that we do need to program -- but as far as I know there is no other way to do what we want.  Happy to be wrong, I just couldn't figure it out at the time.

PR #409 (2023-07-31): Check auxflash in `humility flash`
This adds an auxflash check to `humility flash`:

- `humility flash` will now reflash if the image matches but the auxflash is not loaded
  - This is a _little_ inefficient, because it will also re-flash the main firmware, but was easiest to implement
- `humility check` will now return an error if the auxflash is not loaded

We never actually read back the auxflash; we verify the Hubris image, and the Hubris image contains a checksum for the auxflash.  If the auxflash was invalid, then we would not see an active slot in the running Hubris image.

Fixes #408
closed 2023-07-31 by mkeeter