oxidecomputer/humility #417: Repl enhancements

lzrd, 2023-09-05:
It would be useful if the repl could be used in scripting and/or provided scripting.

  - [ ] Input from stdin (not just /dev/tty) should be accepted
  - [ ] Comments and blank lines should be ignored
  - [ ] All commands should respect the `-j` flag for JSON or other parser friendly output format.
  - [ ] Ensure that humility commands' success/failure status can be parsed.
  - [ ] Doing more than the above quickly gets to the point of embedding an interpreter, evaluate that effort and decide what to do.

This should work but doesn't because `reedline` forces input to come from /dev/tty and not stdin.
```
egrep -v '(^#|^$)' <<! | humility ...
# Get task status:
tasks

# Comments and blank lines should be ignored without filtering input.
!

steveklabnik, 2023-09-05:
Re the reedline bug: https://github.com/nushell/reedline/issues/509

jclulow, 2023-09-05:
It would be pretty neat to be able to pass a [rhai](https://rhai.rs/) program to `humility` that was able to run commands, inspect their output, make decisions, cause the `humility` process to exit with a non-zero status code, etc.  I suspect we could use that in manufacturing for things that we, today, perform with `bash` scripts that chain multiple `humility` invocations, etc.  It could also be useful in a future series of automated dump classifiers.