oxidecomputer/humility #530: Updating probe-rs requires USB work

labbott, 2025-01-15:
probe-rs switched from a `libusb` based backend to `nusb` (https://docs.rs/nusb/latest/nusb/). This unfortunately lacks illumos support at the moment. This means we are stuck on our old version of probe-rs until we add that support.

From discussions with our local illumos experts this seems doable (possibly using something like https://illumos.org/man/4D/ugen)

labbott, 2025-01-15:
I think we still need a custom hidapi for illumos which will probably need a bump as well

thenewwazoo, 2025-08-25:
Hi! 👋 

I stumbled across this while working on updating humility to probe-rs 0.29, which added support for the RP235x, as part of my porting Hubris to the Pico 2 (aka RP235x). I'm just checking in to see if there's been any news. I'm not an Illumos expert (or even user, tbh), so I'm unfamiliar with the required work here, but I suspect updating Oxide's fork of probe-rs to a version that doesn't run on Illumos is going to be a nonstarter.

thenewwazoo, 2025-09-19:
FYI I just submitted a PR to [update probe-rs to nusb 0.2.0](https://github.com/probe-rs/probe-rs/pull/3562).

Getting probe-rs updated in a pre-req to submitting my port of nusb to illumos. :)