oxidecomputer/humility #547: Segfault on illumos with more than 3 mculink devices attached

luqmana, 2025-05-21:
```
$ humility probe
[1]    392 segmentation fault (core dumped)  humility probe
```

```
$ mdb core
Loading modules: [ libumem.so.1 libc.so.1 ld.so.1 ]
> $G
C++ symbol demangling enabled
> $C
fffffc7fffdf2b00 libusb-1.0.so.0.3.0`log_v+0x5e(30200000409fa, 4, fffffc7fef38d1e0, fffffc7fef38c99a, fffffc7fffdf2b38)
fffffc7fffdf2c20 libusb-1.0.so.0.3.0`usbi_log+0xbc(30200000409fa, 4, fffffc7fef38d1e0, fffffc7fef38c99a)
fffffc7fffdf2cb0 libusb-1.0.so.0.3.0`illumos_get_active_config_descriptor+0x215(3b21f20, 3b21f20, 8b)
fffffc7fffdf2d10 libusb-1.0.so.0.3.0`get_active_config_descriptor+0x3b(3b21f20, 3b21f20, 8b)
fffffc7fffdf2d60 libusb-1.0.so.0.3.0`libusb_get_active_config_descriptor+0x7f(3b21f20, fffffc7fffdf2dd8)
fffffc7fffdf2e30 hid_enumerate+0xfc()
fffffc7fffdf3030 hidapi::HidApi::get_hid_device_info_vector::h51a31bfc6108660c+0x56()
fffffc7fffdf3100 hidapi::HidApi::new::h810dc4c22ddd022a+0x32()
fffffc7fffdf32a0 probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::h39a8927e5807e4c6+0x11f()
fffffc7fffdf3340 probe_rs::probe::Probe::list_all::had74725d3e179ab9+0x21()
fffffc7fffdf3440 humility_probes_core::get_usb_probe::h48b2d7a15df079c8+0x26()
fffffc7fffdf39b0 humility_probes_core::attach_to_chip::hfe86a511866698a2+0x965()
fffffc7fffdf3f20 humility_probes_core::attach_to_chip::hfe86a511866698a2+0xee3()
fffffc7fffdf3f70 humility_probes_core::attach::h39807b829f892e20+0x49()
fffffc7fffdf3fc0 humility_cmd::attach_live::ha8455e39636baa97+0x91()
fffffc7fffdf4030 humility_cmd::attach::hdecd3058884e0371+0x67()
fffffc7fffdf4ef0 humility::cmd::subcommand::ha8ffd550988b7f72+0x4cc()
fffffc7fffdf6490 humility::main::hdb6975b3b2eba7fb+0xa63()
fffffc7fffdf64a0 std::sys::backtrace::__rust_begin_short_backtrace::h50c471f8ffacda62+6()
fffffc7fffdf6500 std::rt::lang_start::_$u7b$$u7b$closure$u7d$$u7d$::hbd21f126a1ae0339 +0x11()
fffffc7fffdf6580 std::rt::lang_start_internal::h897570f42d89b337+0x1fc()
fffffc7fffdf65a0 main+0x33()
fffffc7fffdf65d0 _start_crt+0x87()
fffffc7fffdf65e0 _start+0x18()
```

jclulow, 2025-05-21:
@luqmana What system is this on?  It's probably an issue with our libusb package.

jclulow, 2025-05-21:
Ah, I infer from chat messages it is **jeeves**, which I think probably has not been updated in quite a while seeing as it did not have any USB devices until yesterday.  I have updated **libusb**:

```
jclulow@jeeves ~
 $ pfexec pkg update -v libusb
            Packages to update:         1
     Estimated space available:   2.72 GB
Estimated space to be consumed: 357.11 MB
       Create boot environment:        No
Create backup boot environment:       Yes
          Rebuild boot archive:        No

Changed packages:
helios-dev
  library/libusb
    1.0.25-2.0 -> 1.0.25-2.1

DOWNLOAD                                PKGS         FILES    XFER (MB)   SPEED
Completed                                1/1           2/2      0.2/0.2  1.2M/s

PHASE                                          ITEMS
Removing old actions                             1/1
Installing new actions                           1/1
Updating modified actions                        3/3
Updating package state database                 Done
Updating package cache                           1/1
Updating image state                            Done
Creating fast lookup database                   Done
Reading search index                            Done
Updating search index                            1/1
Updating package cache                           2/2
```

Can you try again?

luqmana, 2025-05-21:
That seems to have fixed it, thanks!

closed 2025-05-21 by luqmana

labbott, 2025-05-22:
The dream is still to finish the `nusb` port (https://github.com/oxidecomputer/nusb/tree/illumos) so that we can use that instead of `libusb` and an upgraded probe-rs