oxidecomputer/humility #548: Humility takes excessive time to execute command on Windows

ericaasen, 2025-05-21:
When using Humility with a single USB probe, it takes many seconds between when a command is started and when it actually executes.

The only external USB things on my system:

Adafruit FT232H dongle for a UART
Keychron K10 Keyboard
ST-Link V3

System Info:
```
Device name	LAPTOP-UL0DMADE
Processor	Intel(R) Xeon(R) W-10855M CPU @ 2.80GHz   2.81 GHz
Installed RAM	64.0 GB (63.7 GB usable)
Device ID	8EB994C6-E03F-4A47-ADCD-5E07EE8A161A
Product ID	00391-50007-08830-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Pen and touch support with 10 touch points

Edition	Windows 11 Pro for Workstations
Version	24H2
Installed on	‎3/‎2/‎2025
OS build	26100.4061
Serial number	R912JB1J
Experience	Windows Feature Experience Pack 1000.26100.84.0
```

https://github.com/user-attachments/assets/e423bb14-6811-4722-9fff-683e1f4c8b05

sdonnan, 2025-08-21:
I can reproduce this with `probe-rs` command line tools as well.

I have a attached a log of the behavior.

[download.log.gz](https://github.com/user-attachments/files/21926082/download.log.gz)

sdonnan, 2025-08-22:
The delay has disappeared for me. Unfortunately I did a bunch of things to my machine before testing again.

- Did https://github.com/oxidecomputer/meta/blob/master/engineering/hardware_engineering/wsl.adoc (didn't retest)
- Put computer to sleep (didn't retest)
- Tried `cargo flamegraph --bin humility --release -- probe` got permissions issues
- `cargo flamegraph ...` as admin - NO DELAY
- Regular cargo run as admin - NO DELAY
- Regular cargo run as user - NO DELAY
- "unbound" from usbip-win - NO DELAY
- Rebooted - NO DELAY
- uninstalled `usbpid-win` and `wsl-usb-manager` - NO DELAY

I have not been able to reproduce the delay after these steps.