oxidecomputer/humility #569: Update open pin and phase check for Cosmo

ericaasen, 2025-09-25:
It seems like the RAA229620 has not been added to `rendmp` for cosmo based on this error message:

```
λ humility rendmp --open-pin --device 0x75
humility: attached via ST-Link V3
humility rendmp failed: no RAA229618 or ISL68224 with address 117; use `humility pmbus -l` to list devices
```

Also, when I try to run it on the ISL68224 on cosmo at address `0x5c` it times out
```
λ humility rendmp --open-pin --device 0x5c
humility: attached via ST-Link V3
humility rendmp failed: operation timed out
```

nathanaelhuffman, 2025-12-01:
@bcantrill @mkeeter Note that lacking this functionality prevents us from getting better diagnostics to benchmark other than "check controller and phases" (see https://github.com/oxidecomputer/mfg-troubleshooting/issues/164 as an example).  Using this on gimlet has let the techs immediately isolate the problematic phase, so we'd like to support this in cosmo asap also as we ramp mfg.

cbiffle, 2026-02-05:
Here's what I'm currently getting from phase check. Note that the machine is in A2, and since I'm using a headless A-rev without a CPU I don't think I can get it into A0. Should I find a CPU'd machine to test on?

```
Phase check for RAA229620A at 0x75
PHASE |    VOUT    |    IOUT    |    TEMP    | RAIL
------|------------|------------|------------|------------
 0    |     0.000V |     0.000A |  341.000°C | VDDCR_CPU0_A0
 1    |     0.000V |     0.000A |  341.000°C | VDDCR_CPU0_A0
 2    |     0.000V |     0.000A |  341.000°C | VDDCR_CPU0_A0
 3    |     0.000V |     0.000A |  341.000°C | VDDCR_CPU0_A0
 4    |     0.000V |     0.000A |  341.000°C | VDDCR_CPU0_A0
 5    |     0.000V |     0.000A |  341.000°C | VDDCR_CPU0_A0
 6    |     0.000V |     0.000A |  341.000°C | VDDCR_CPU0_A0
 7    |     0.000V |     0.000A |  341.000°C | VDDCR_CPU0_A0
 8    |     0.000V |     0.000A |  340.000°C | VDDCR_SOC_A0
 9    |     0.000V |     0.000A |  341.000°C | VDDCR_SOC_A0
 10   |     0.000V |     0.000A |  341.000°C | VDDCR_SOC_A0
 11   |     0.000V |     0.000A |  341.000°C | VDDCR_SOC_A0
humility: restoring default configuration...
humility: done restoring default configuration...
```

cbiffle, 2026-02-05:
This should work if you update Humility _and_ Hubris, because there turned out to be a firmware change required.

The firmware change _is not_ in the current R18 candidate.